AlanChatham/UnoJoy #3: Arduino Uno Bricked

gareththegeek, 2015-07-26:
After following the instructions, putting the Arduino Uno r3 into DFU mode and running the batch file, I receive an error.  After this I was unable to get the chip back out of DFU mode and ended up throwing the chip away.

I think you should make it more clear, what the risks are for following your instructions.

NicoHood, 2015-07-26:
You cannot brick an Arduino with uploading a new firmware. If you are able to delete it you are also able to upload a new one. If not your Arduino was crap before or you did something wrong/read not the tutorial carefully.

You can still burn the whole dfu + usbserial again with another Arduino. Or you upload the HoodLoader2 which is better anyways. Then you can also switch between firmware and USB-Serial faster. Both bootloaders can be burned with the installation sketch:
https://github.com/NicoHood/HoodLoader2

0ddBall, 2015-08-27:
I have the same bored and thought I had the same issue, but instead of blaming this project I spent 6 months off and on trying to recover it. The problem was simply the DFU programmer being outdated and not working. I found a new version flash the Arduino bootloader and I was good to go.

http://sourceforge.net/projects/dfu-programmer/?source=dlp

moeburn, 2017-01-14:
Can confirm.  Download dfu-programmer-win-0.7.2.zip from 0ddBall's link, then go into the dfu-prog-usb-1.2.2 folder inside the zip, then copy those contents to UnoJoy-master\Drivers\Windows, then you can run UnoJoyDriverInstaller.bat.

If you try to run UnoJoyDriverInstaller.bat _without_ downloading the updated DFU programmer, you'll get a red X on one of the two drivers the UnoJoyDriverInstaller.bat program tries to install.  With the updated DFU programmer, you get two green checkmarks.

Maybe it's just a Windows 8.1 and higher thing, who knows.

closed 2017-11-17 by gareththegeek