AlanChatham/UnoJoy #4: Cannot run the visualizer

aggiehavoc, 2016-01-05:
When I try to run the processing visualizer, i get this error:

>Could not find the main class.  Program will exit.

I am thinking that this might be a JRE version issue or something.  Have you seen this before?  I am running the latest version of UnoJoy here with JRE v. 1.8.0_66.  I have tried running the standalone EXE, as well as the jar by itself.  Both give me the same error.  I am not super fluent in Java, so it might be a config issue.  Any help would be appreciated.

AlanChatham, 2016-01-05:
There have always been a bunch of problems with the executable versions of the visualizer programs - the best way to do it is to just download Processing from processing.org and running the .pde file.

aggiehavoc, 2016-01-06:
OK.  Thank you!

sikokinesis, 2016-01-08:
I downloaded Processing 2.2.1. It wont work with Processing 3

hightowersoft, 2016-06-17:
I downloaded 2.2.1 and I can get the visualiser to load now, but it doesn't change when I press a button. Uno works fine in joystick mode though, just can't see any button presses in the visualiser in arduino mode.

alfredomova, 2019-02-06:
update to the current jre available,