AlanChatham/UnoJoy #6: Can't load jvm.dll

WORMSS, 2016-01-14:
Windows 7:
When running "TurnIntoAJoystick.bat" I get "Can't load jvm.dll"

Did I install Atmel Flip Program? **Yes**
Is the arduino plugged in? **Yes**
Is it in 'Arduino Uno DFU' mode? **Yes**

I tried symlinking the jvm.dll in from several location TO several locations.
First it was in the same folder as `TurnIntoAJoystick.bat` , **but nothing**
I then found where `batchisp` was installed and put it there, **nothing.**
I then placed it in ATmega8u2Code **nothing**
Placed `C:\Program Files\Java\jdk1.7.0_45\jre\bin\server` on %PATH%, **nothing**

So now I am afraid I am at an end to the tricks I can think of.. If anyone else can help?

- Colin

Sodierlow, 2017-01-07:
Hey, You need to install Java JDK

JamesSanii, 2017-01-08:
I also have the "Can't load jvm.dll" error message. 
I installed both the 64 and 32 bit versions of the Java 8u111 SDK.
I did note that my path variable has: C:\ProgramData\Oracle\Java\javapath
and the items in javapath are symlinks to the 64Bit version C:\Program Files\Java\jre1.8.0_111\bin
Do I need to be pointing to the 32bit version ?

rodrirochad, 2017-04-17:
I have the same error. I am getting really frustrated with this. Now I can't even get back the arduino working properly.

Martinnygaard, 2017-05-01:
Same error here. I would appreciate some help here!

cvillegaslobos, 2018-01-27:
For thos who has this problem, I just solved it following the steps in this link [https://www.java.com/en/download/help/error_mainclass.xml](url)

redddcyclone, 2023-09-01:
@IonicOwl @WORMSS 

The problem is that batchisp.exe requires Java 5. You can fix this by installing the version of Atmel Flip that comes with the JRE included.