AlanChatham/UnoJoy #9: increase resolution

paulelsner, 2016-11-17:
Hi,
I use UnoJoy for connecting my model transmitter with my PC to use as controller for flight simulators. This works great, but the resolution could be improved. 
Is it possible to extend the value range to 16bit? I have an input resolution of 1000 values.

Thanks!
Paul

AlanChatham, 2016-11-20:
You should take a look at the MegaJoy! variant firmware - it's been like half a decade since I wrote it, but a brief look suggests that it's got 16 bit resolution for it's axes. The UnoJoy! firmware is specifically designed to imitate a Playstation 3 controller as close as possible, so that's why it has 8 bit axes. So MegaJoy! would be a great place to start, and if that gives you issues, comparing the code between the two should at least get you a good start if you wanted to modify UnoJoy! to extend the axes.

paulelsner, 2016-11-20:
Thanks! I think I know what I have to change. But I don't know how to recreate the hex file for flashing the controller. Is that possible with the Arduino suite?

AlanChatham, 2016-11-22:
I don't believe it's possible with Arduino. All the hex file stuff was developed with Atmel Studio, and so I would use that, since the project files are all there so all the code should be all set up to work relatively straighforwardly...

paulelsner, 2016-12-10:
I found some time to look into that and found a solution. Now I have a version of UnoJoy available where the two analog sticks have 10bit resolution.
I uploaded my Arduino code for reading a PWM receiver and the according  .hex file to my dopbox: https://www.dropbox.com/sh/4fxa55napu0leg2/AAAlarZvOTDvEpadoomsMBexa?dl=0
You'll also find the changed code there.

closed 2016-12-10 by paulelsner