AlanChatham/UnoJoy #10: Arduino/Genuino 101 support

fedebecat, 2016-11-19:
Hi, thanks for the great work, is there any chance this may be compatible with an Arduino/Genuino 101? It would be very cool to use the sensors as inputs for a gamepad. Any plans in adding a support for this board?
Thanks!

AlanChatham, 2016-11-20:
Hm, the chipset on the Genuino 101 is totally different, so writing UnoJoy! software for it would involve writing an entire new set of firmware, so I don't personally have any plans to support this board. One thing you could do would be to take the UnoJoy.h file, modify it so it was using the Genuino 101's hardware serial (the Serial1 object, by the looks of it), then you could run a wire from the TX pin on the Genuino to the TX pin on an Uno that's been flashed with the UnoJoy! firmware? That way, the Genuino would send out the serial signals that the communications chip on the Uno knows what to do with.

fedebecat, 2016-11-20:
That's a good idea, I'll give it a try! Thanks! :)

closed 2019-03-19 by AlanChatham