AlanChatham/UnoJoy #11: Installer not working on MacOs Sierra

marbaque, 2017-01-06:
Hi, I was trying to install the driver on mac, but it is only for older versions. Is there a workarounf or another version coming soon?

shemit, 2017-02-21:
I actually got the YosemiteUnoJoyDrivers.pkg installer to work. However, I'm running into issues using the Processing visualizer due to controlP5 compatibility issues with Processing 3. I'll try to see what happens if I use an older version of Processing.

ubercast, 2018-02-11:
I've got the same moment. @shemit , how you've managed to user Yosemite driver?

shemit, 2018-02-11:
It's been a while, so I tried running the drivers installer just now and that worked fine. It looks like the real issue is that the controlP5 library is outdated. IIRC, I simply skipped visualization, and went straight to working on the arduino and prototyping there.