AlanChatham/UnoJoy #17: DPAD buttons behave incorrect on PS3 (fix)

phaseIV, 2017-06-29:
I have succesfully connected my old Negcon controller to the PS3 with Unojoy, so many thanks for making that possible! However, the (DPAD) directional buttons were all wrong in the game menu (Wipeout HD) while they were working fine in the PS3 main menu.

Looking further into this issue I found that the PS3 expects the DPAD buttons to return a digital value as well as an analogue (sensitivy) value. It seems that Unojoy interprets these analogue values incorrect, as you might suspect when you read the following code snippet. :-)

	// Gonna assume these are button analog values for the d-pad.
	// NOTE: NOT EVEN SURE THIS IS RIGHT, OR IN THE CORRECT ORDER
	uint8_t up_axis;
	uint8_t right_axis;
	uint8_t down_axis;
	uint8_t left_axis;
So I rearranged the lines in [usb_gamepad.h](https://github.com/AlanChatham/UnoJoy/blob/master/UnoJoy/ATmega8u2Code/usb_gamepad.h#L56) as shown below, built a new hexfile and now the DPAD buttons behave correctly in the Wipeout game menu too.

	uint8_t right_axis;
	uint8_t left_axis;
	uint8_t up_axis;
	uint8_t down_axis;

Hopefully this helps anybody, I'm back to racing! Thanks again.

HollyBrius, 2017-11-18:
Hi PhaseIV
Thanks for your solution, I have the same pb on FUEL game.
When you said "built a new hexfile", how do you do that ?
Tx
H.

phaseIV, 2017-11-20:
To create a new hexfile you need to recompile the sourcecode from the Unojoy/ATmega8u2Code directory with the AVR Studio 5.1 IDE/compiler (Windows only). Or you could try and use my Unojoy hexfile which you can download [here](https://www.dropbox.com/s/tayrijb8pvpn25o/UnoJoy.hex.zip?dl=0). Hope it works for you!

phaseIV, 2017-11-21:
Glad to hear that the hexfile was useful. Good luck with the project!

HollyBrius, 2017-11-21:
Thanks !
It would be great if AlanChatham could change his Unojoy :)

damccull, 2018-02-20:
@hollybrius, best way to get a change into open source code is to fork it, change it on yours, then make a pull request to the original repo.