AlanChatham/UnoJoy #19: How do you compile a new hex bootloader file?

TheRoboman, 2017-12-31:
Hi Alan,

I am using your code to try to interface with the Original Xbox. I have modified the usb_gamepad.c file, but I don't know how to compile into a hex bootloader file to test it. 

Can you please recommend how I should go about compiling?

TheRoboman, 2018-01-01:
Managed to figure it out! Install Atmel Studio, open the provided atsln solution file. Then go to Build -> Configuration Manager. Change the Configuration for the UnoJoy project to HexFiles. Then just build the project when you are done editing it and it will generate the hex files for you to flash.

waylander131, 2018-07-27:
one question: after modifying usb_gamepad.c and usb_gamepad.h do I need to compile a new hex file?
an other words how are those modifications implemented in the Arduino Uno board? Are they reflected in the microcontroler or the 8u2 chip code?

AlanChatham, 2018-07-28:
You'll need to compile a new hex file for anything outside of the UnoJoy.h file that gets included in your Arduino project. Hope that helps!

waylander131, 2018-07-28:
Thanks Alan, it helps.
I still struggle with the descriptor though, I'm trying to create simple joystick with 6 analog axis only, but aparently I can't figure out the descriptor - I get no data coming from the joystick although it is correctly recognised. Is there any guide out there that could help me?

AlanChatham, 2018-08-01:
Ha, yeah, that's a big jump. HID report descriptors are a real bear, and debugging them is a lot of frustrating trial and error. One tip would be to make sure you're uninstalling the device each time before testing new code with it - if you don't, at least Windows will use a cached copy of the descriptor or something, and things won't work, or at least that's what I remember from 5 years ago.
If you're looking for more axes, though, I might suggest just using the MegaJoy code - that's got a bunch of axes for use.