AlanChatham/UnoJoy #25: Raising the amount of buttons on megajoy

Mikkelhost, 2019-01-01:
Hi

Im trying to make som flightsim hardware as cheaply as possible. So i started using this, and it works great with 64 buttons. But when i edit the descriptor and everything to add more buttons. as an example im using 80 buttons for this project. The buttons 64-71 (8 buttons which is exactly 1 byte of buttons) doesnt work. i have looked through the descriptor and everything, and i can't find out why this 1 byte of buttons doesn't work.

Mikkelhost, 2019-01-01:
[image]
Here is a gif of what im talking about. I made the arduino cycle through the buttons from 0-79 and then repeat. As seen on the gif, every button works exept those 8 buttons. 

I also tried one time changing the amount of buttons to 144, her there are 16 buttons that doesn't work. more specifically buttons 64-71 and 128-135. So it seems to behappening at every 8th byte.

AlanChatham, 2019-01-08:
Hey, thanks for looking into this! I have ZERO idea why that is the case,
and it's been like 8 years or whatever since I really was doing a bunch of
USB stuff, so I'm not gonna really be any help, unfortunately.... I do seem
to recall I set MegaJoy up with 64 buttons originally since I couldn't get
a higher number of buttons to work in Windows, so I guess I just didn't go
high enough lol.

My recollection is that programming USB descriptors was a real ugly mess,
since I didn't ever learn USB too deep, so my recommendation is to maybe
just map your buttons in your sim to step around the dead block of buttons?
Not the most elegant solution, but you may be in a niche case. Might be
interesting to post a bug report on the GitHub about it and see if anyone
else eventually pokes around on it?

Thanks for putting the time into messing with it, and sorry I'm not much
help...

-Chatham

Mikkelhost, 2019-01-08:
I've fixed the problem. Found out that it was in the arduino ". h" file itself. Don't know why i hadn't looked there yet😅 but seems to have fixed the problem. Now i would just have to find out how i will be able to activate all buttons at once. Maybe you have an idea on how that is supposed to happen :-)

Best regards
Mikkel

AlanChatham, 2019-01-10:
If you wanted to activate all the buttons at once, shouldn't you be able to
set up the dataForController_t that you're using to have all the buttons
on? Let me know if that's not working!

Thanks,
-Chatham

Mikkelhost, 2019-01-10:
Im not sure i understand quite what you mean. All i have tried is to let the arduino run through a sequence which turns all the buttons on. but it doesnt work :/

AlanChatham, 2019-01-10:
I mean something like this from the sample MegaJoy code:

...

megaJoyControllerData_t getControllerData(void){

  // Set up a place for our controller data
  //  Use the getBlankDataForController() function, since
  //  just declaring a fresh dataForController_t tends
  //  to get you one filled with junk from other, random
  //  values that were in those memory locations before
  megaJoyControllerData_t controllerData = getBlankDataForMegaController();
  // Since our buttons are all held high and
  //  pulled low when pressed, we use the "!"
  //  operator to invert the readings from the pins

  // old line
  //for (int i = 2; i < 54; i++){
  //  controllerData.buttonArray[(i - 2) / 8] |= (!digitalRead(i)) << ((i -
2) % 8);
  //}
  //new line, for however many buttons, assuming you also changed what a
dataForController_t struct looks like
  // to have enough size for the buttons  here like: #define
BUTTON_ARRAY_SIZE 10
  for (int i = 2; i < 80; i++){
    controllerData.buttonArray[(i - 2) / 8] |= (0b1) << ((i - 2) % 8);
  }

.....

Let me know if that makes sense? It's using bitwise operators in the
MegaJoy! code to toggle button states, as MegaJoy holds its button data as
an array of bytes, with each byte holding the button data for 8 buttons,
one for each bit.

Hope that helps!
-Chatham

Coachidoachi, 2019-01-21:
Hello there, 
at the moment i plan on building a similar Project for my flightsim.
I am planning for ca. 52 digital inputs.
However, i would like to build it, so i can expand on it in the future.
My problem is, no matter what i do, Pointys Joysticktester only shows 2 devices with 32 buttons.
I veryfied this by trying to assain a button beyond that to a controll in the flightsim (DCS) and it doesn`t recognize it either, so something isn`t working.
I don´t have that much experience in programming and i tried solving this problem for the last 3 days, so i thought that someone here could maybe help me.
What i did was changing the button array size in Megajoy.h and expanding the "for loop" to assain the new buttons in the array.
Can someone please give me a hint on what to try additionaly?
I am currently testing running on a 64bit Win10 computer.
Thank you and greetings!
(and sorry for my bad english speaking skills, i am working on them :) )

Mikkelhost, 2019-01-21:
Hey Coach

add me on discord 
Mikkelhost#1366

I got something that might help you out

briancavanaugh, 2019-02-09:
@AlanChatham @Mikkelhost I was going to start a new issue thread, but I saw this one and based on what I see as the original post, I believe this ties directly to what I am trying to do.  I'm trying to use UnoJoy to add a couple more buttons but I'm having a bit of difficulty.  For starters, I've only tried to add 1 new button as a test to see what I get.  I have download Atmel Studio 7, but I'll admit this is the first I've used it, so I'm not exactly sure what I could be missing.

For starters, I modified the UnoJoyArduinoSample and UnoJoy.h files and flashed them to my UNO.  For my test, I decided to make pin A3 a digitalWrite, just like A4 and A5.  After turning into a joystick, obviously nothing new happened because I didn't change any of the ATmega code.  So........

Next, I modified the following files in the ATmega8u2Code folder...
usb_gamepad.h
`uint8_t newTest_btn : 1;`
usb_gamepad.c
`.newTest_btn = 0` (added in the gamepad_idle_state function signature)
`gamepad_state.newTest_btn = btnList.newTestOn;` (in the sendPS3Data function)
UnoJoy.c
`dataToSend.newTestOn = 1 & (buttonData3 >> 1);`
dataForController_t.h
`uint8_t newTestOn : 1;` (right below the dpadDownOn line)
ButtonMapping.txt (I realize this is only a reference file)
physicalButtonList_t.h (this file didn't seem to be included in the solution, so after my first attempt which didn't work, I decided to include it, but got the same results)

After modifying and building successfully, I copied the output files into the ATmega8u2Code/HexFiles folder and turned into a joystick again.  Unfortunately, I did not get a new button to show up in the controller tester.   However, this time grounding the A3 pin did cause button 5 to turn on.  I'm really unsure why it made that button turn on.

Is there some sort of constant I need to increase that defines the total number of buttons?  I can include additional code snippets if I've been unclear on anything.  I appreciate any help with this.

AlanChatham, 2019-02-09:
Are you using UnoJoy or MegaJoy? MegaJoy has a bunch of buttons, way more than the 13 that UnoJoy has, so if you haven't, give that a shot.
Otherwise, you're going to be entering the horrible world that is USB HID device programming, which is a real pain. If you're trying to add more buttons, you'll also need to update the HID report descriptor, and possibly some other HID stuff somewhere (I don't totally remember at this point, but doing a diff between the UnoJoy and MegaJoy files could help a lot there).
Importantly, between tests, you'll need to uninstall your controller from Windows, since the OS will cache the HID report descriptor between device plug-ins, so if you don't uninstall the device on the system level, it won't actually pay attention to the new HID device report you're trying to give it.

Hope this is helpful!

briancavanaugh, 2019-02-09:
Thanks for such a quick reply!  I found the issue though!  It was in the gamepad_hid_report_desc constant.  I saw that I had to increase the REPORT_COUNT and USAGE_MAXIMUM, but missed the fact that I then had to decrease the REPORT_COUNT for the next INPUT which was essentially the padding.

Also, I found the other issue that I had of the wrong button turning on from my pin.  For that one, I simply had the order of my button names wrong in the typedef struct in my usb_gamepad.h file.

FYI....I'm using UnoJoy, not MegaJoy.  The buttons that I'm adding, I'll be able to fit in using a matrix pattern with the number of pins that the UNO provides, plus I found a couple of UNOs for much cheaper than the Megas.  I'm learning a ton with this in a very short while.  Thanks for creating a cool project!

AlanChatham, 2019-02-11:
Glad things worked out! Just FYI, you should be able to use the MegaJoy! code on an Uno, giving you more buttons to play with.

briancavanaugh, 2019-02-12:
Thanks!  I'll give that a shot too.

fihnakis, 2019-08-24:
> Hey Coach
> 
> add me on discord
> Mikkelhost#1366
> 
> I got something that might help you out

Mikkelhost, I have a project I'm working on similiar to the one you are talking about. I sent you a friend request on discord (fihnakis) if that's ok to discuss it. I'm building a 66 button game controller with 2 3-axis joysticks but since the MEGA only has 54 pins to use for button inputs I could use some help to get the amount of buttons increased as you are talking about.

Assamita81, 2023-08-23:
Hello mr @Mikkelhost. I wanted to ask you about your project because I'm doing something similar, but couldn't find your Discord ID. Would you be so kind to contact me for some help? my ID is SrSosio#4606
Thank you

Mikkelhost, 2023-08-23:
Hi @Assamita81 I abandoned my own project years ago.
So im no longer able to assist as i havn't looked into it since 😕

Mikkelhost, 2023-08-24:
I believe that i ended up using a matrix yes, and then i used diodes to make sure that i could push multiple buttons at the same time.

cristiano1864, 2024-04-27:
Mikkelhost, Could you please share your code with the increased number of buttons ?
I'm trying to make two devices with 64 buttons and I changed them 
in MegaJoh.j
#define BUTTON_ARRAY_SIZE 16
in usb_gamepad.c
0x95, 16 * BUTTON_ARRAY_LENGTH,        //   REPORT_COUNT (64)
0x29, 16 * BUTTON_ARRAY_LENGTH,        //   USAGE_MAXIMUM (Button 64)
- static const gamepad_state_t PROGMEM gamepad_0_idle_state = {
.buttonArray = {0,0,0,0,0,0,0,0},
static const gamepad_state_t PROGMEM gamepad_1_idle_state = {
.buttonArray = {0,0,0,0,0,0,0,0},
in MegaJoyArduinoSample.ino
for (int i = 2; i < 128; i++){
    controllerData.buttonArray[(i - 2) / 8] |= (0b1) << ((i - 2) % 8);
and the board sees an increase in the number of buttons, but none of them work.
I don't know what I'm doing wrong.