AlanChatham/UnoJoy #29: MegaJoy with Seeed Can bus shield

hu33ard, 2019-02-28:
I'm using a MegaJoy, I have it operating just fine as a joystick using analogue inputs with X360ce as my emulator.

For the project that I am working on though, I need to do more with the signals generated by the analogue joystick. So I have implemented the Seeed CAN bus shield to allow comms between another plc and the arduino. The communication works great, until I turn the mega in to joystick. The signals being processed by the emulator become far-far away from the inputs that I am sending to the analogAxisArray...Any particular reason you know of as to why this wouldn't work?

AlanChatham, 2019-02-28:
Am I understanding this correctly : you switch to a joystick, and the PC recognizes it as a joystick, but the values it's showing don't match up with what you expect to see from whatever input you're sending over your CAN bus?

My best guess would be that you might be having an underlying issue with the way that MegaJoy is using timer interrupts to handle the way it talks back and forth to the USB chip on the board? Like, the interrupts it's using are messing with your CAN bus reads possibly?

hu33ard, 2019-03-01:
Yes that's a good summary. When I did the initial testing with the Arduino in the normal mode. using the serial monitor I can see that all the CAN signals are being read correctly by the board, but after switching to a joystick the values don't make any sense in the controller emulator that's reading the data. 

Yeah that sounds like a potential answer, do you know which file(s) the interrupts initialised in so I can take a look? Or do you think, as I am beginning to think, that maybe this combination of hardware/software is incompatible...

AlanChatham, 2019-03-01:
You'll find the timer stuff in MegaJoy.h