AlanChatham/UnoJoy #35: Logitech G2x shifter interfacing with MegaJoy

MineAngerBR, 2020-07-07:
Hello, i'm trying to interface analog-based H-shifters, just like the common and available shifters from Logitech for the G series (G25, G27, and Logitech Driving Force Shifter for the G29/G920)
I'm not a programmer in any means or stretch, just one enthusiast wanting to build something more integrated, so i don't need to have multiple boards for each part of the simulation mechanisms.
With some code from the youtuber AMStudio, i got the basics of the interface (it actually works, changing the position of the analog axis activates the button), but it lags a lot (from 1,5 up to 5 seconds between shifts), while the other buttons work just as fast as they always worked, i would like to know how can i optimize it to shift faster

the file is for MegaJoy: 
[MegaJoyArduinoG2xShifter.txt](https://github.com/AlanChatham/UnoJoy/files/4886215/MegaJoyArduinoG2xShifter.txt)

AlanChatham, 2020-07-08:
Hi!

Hmmm, if I had to guess, I might think that setting the input_pullup
resistors on on your analog pins might be screwing up your analog reads?

My first suggestion would be to try a sketch that just outputs the values
of your analog reads to the serial port without MegaJoy, to see if you're
getting the sort of quick response on your shifter that you'd expect, so
you can figure out if it's more of a problem with the electronics or the
MegaJoy interface.

Hope this is helpful!
-Chatham

MineAngerBR, 2020-07-08:
Hey Alan, thanks for answering me.

Yeah, i made a small sketch to test my analog reads and also to output as the gear selected, and they're being quick and responsive. And about the pullup resistors, i've removed that bit of code from the original sketch to test and the responses are still slow.

AlanChatham, 2020-07-08:
Hm, the only thing that jumps out to me is the use of delay() in the code -
I'm thinking that the way MegaJoy uses timers and interrupts may be
interfering with those in a way that makes them take much longer? Try
removing them and see what happens. If you're having issues with
flickering, you might, instead of a delay, compare against a global value
to make the code compare the shifter value across a couple of loops to make
sure it's settled in a new position.

Hope it helps!

MineAngerBR, 2020-07-08:
I removed the delay() to test, still no difference.
The problem in MegaJoy isn't flickering, is the response time between moving the position in the XY axis, and the arduino "pressing" the correct button.
All the buttons are okay and responding quickly, even the digital pins 3-8 when i pull them to GND with a jumper.
Just the "XY-to-button" conversion is problematic.

AlanChatham, 2020-07-08:
Hm. I'll poke around later today and try out your code - seems like an
interesting puzzle....

MineAngerBR, 2020-08-05:
Alright, i tried to optimize it messing with the "if-else" statements, but no change in the speed of the responses from the "XY-to-button" conversion, and the standard way (digital pin and ground) is still ok and working as quick as always