AlanChatham/UnoJoy #36: Issues with Encoder Rotary with MegaJoy

hirakkopk, 2020-09-24:
i'm using MegaJoy, I want to connect two rotary encoders. But for some reason, Megajoy sometimes doesn't read the encoder when I turn it. Has anyone tested an encoder on UnoJoy / MegaJoy?

MineAngerBR, 2020-09-25:
Well, i don't really know what type of encoder are you using, but with help of some code from internet (obviously credited correctly in the head of the sketch) and a library from Paul Stoffregen, i was able to use a quadrature encoder from a printer's paper feed mechanism, and one incremental encoder of the following model: LPD3806-600BM-G5-24C

Just need to rename the extension to .ino to open in the IDE, install the library using the built-in manager (need to be the correct encoder library from the guy i said), and the signal wires from the encoder goes to pins D2 and D3
And don't forget to tell me if it worked for you, crediting as well

[MegaJoyEncoderFinalMod.txt](https://github.com/AlanChatham/UnoJoy/files/5283742/MegaJoyEncoderFinalMod.txt)

hirakkopk, 2020-09-25:
Thanks for answering me!
Your code worked. But I want to activate two digital buttons when
turning the encoder.
Example: Rotate left: Activate D2
Rotate right: Activate D3.

But when I turn it into MegaJoy, only slow spins work. If I try to
rotate quickly, the buttons do not activate accurately.

My encoder: KY-040

Example in video:

hirakkopk, 2020-09-25:
With your code I can adjust it to work for me. However, when i turn the encoder very fast, megajoy does not process data. It only works if you turn it slowly.

Setting "volatile int serialCheckInterval = 1" to "0" in MegaJoy.h helped to decrease the delay. but there is still a little bit.

Do you know how to solve this?

hirakkopk, 2020-09-25:
Video example , rename to .mp4
[encodervideo.txt](https://github.com/AlanChatham/UnoJoy/files/5284594/encodervideo.txt)

MineAngerBR, 2020-09-25:
Unfortunately i won't know how to solve it.
Maybe it is related to the serial frequency, because if i'm not mistaken in the documentation inside MegaJoy.h, line 149, saying:
"Start the serial port at the specific, low-error rate UnoJoy uses. If you want to change the rate, you'll have to change it in the firmware for the ATmega8u2 as well.  250,000 is actually the best rate, but it's not supported on Macs, breaking the processing debugger."

If you know how to compile the firmware for the ATmega16U2, trying a higher frequency won't hurt anything.

MineAngerBR, 2020-09-25:
I am trying to make something similar to you, but instead of pressing keys with an encoder, using two axis to map the position of an h-shifter pattern, but i'm also hitting this delay, and since i'm not that specialist with programming, i don't know how to compile it to try a higher frequency for the serial comm.

hirakkopk, 2020-09-25:
I'm actually quite an amateur in that. But I will try to modify it. I'll let you know if I can.

hirakkopk, 2020-09-25:
Man, managed to change the values ​​in the firmware and it works better.
Tell me which version of UnoJoy you use. Uno / Leo / Mega.

I will send you the modified firmware.

AlanChatham, 2020-09-26:
Hi! I'm guessing that there might be an issue with interrupts - a guess might be that the encoder's use of interrupts is being blocked or delayed by MegaJoy's use of interrupts. Looks like that encoder library has an option for working without interrupts, though - try adding the following line at the top of your code, BEFORE the includes, which should switch the encoder library to not use interrupts:

#define ENCODER_DO_NOT_USE_INTERRUPTS 1

Just a shot in the dark, but I hope it helps!

MineAngerBR, 2020-09-26:
> Man, managed to change the values ​​in the firmware and it works better.
> Tell me which version of UnoJoy you use. Uno / Leo / Mega.
> 
> I will send you the modified firmware.

I use the Mega, but want to use Uno in the future

hirakkopk, 2020-09-26:
Hello AlanChatham! I tried to use your command but it hasn't changed much. The virtual control does not follow the movements of the encoder. Only when I turn slowly.

MineAnger, I changed the serial to 250000. I see if it improves more than that.

MineAngerBR, 2020-09-26:
> Hello AlanChatham! I tried to use your command but it hasn't changed much. The virtual control does not follow the movements of the encoder. Only when I turn slowly.
> 
> MineAnger, I changed the serial to 250000. I see if it improves more than that.

Okay, i'm waiting for you to send me the Mega and Uno firmware please, so i can test here

hirakkopk, 2020-09-26:
Copy and paste into Uno / MegaJoy folders. And tell me if it worked. Don't forget to change Serial.begin to 250000 in UnoJoy.h / MegaJoy.h

Raname for .rar
[UnoJoySerialRateUpdate.txt](https://github.com/AlanChatham/UnoJoy/files/5287081/UnoJoySerialRateUpdate.txt)

AlanChatham, 2020-09-26:
I know this is really kind of the opposite of a solution, but if you don't have any luck with any of this, one way to get around this would be to use a separate Arduino to read the encoder, and you could just send that information to the MegaJoy via the analog pins. Not ideal, clearly, but definitely a way that would work?

hirakkopk, 2020-09-26:
Maybe someone could try this. I only have 1 Mega and I can't buy another one. Where I live is very expensive. Thank you anyway

ouiouiblog, 2021-06-16:
I know that this discussion is a bit old, but if that may help someone.
My encoders were unusable with Unojoy (UNO). I tried the SerialRateUpdate trick bit it didn't.

I found a solution by inserting a delay of 20 ms after the encoder reading (I use encoder.h lib)
`  long NewEncB = EncB.read();
  if (NewEncB != PosEncB ) {
    if (NewEncB > PosEncB ) {
      Hor[3] = 1;
    }
    else {
      Hor[2] = 1;
    }
    PosEncB = NewEncB;
  }
  long NewEncC = EncC.read();
  if (NewEncC != PosEncC ) {
    if (NewEncC > PosEncC ) {
      Hor[5] = 1;
    }
    else {
      Hor[4] = 1;
    }
    PosEncC = NewEncC;
  }
  delay(20);`