AlanChatham/UnoJoy #41: Differing names for Megajoy Devices

X52p, 2021-03-07:
Is it possible to set two different names for the two devices that show up when using Megajoy?
(instead of using UnoJoy Joystick two times, just use UnoJoy Joystick1 and UnoJoy Joystick2)
Some games (especially Unity Games) can't handle it when two devices have the same Name.

Abcdeflijus, 2021-04-03:
I am also interested in that. MegaJoy is amazing thing for flight simulators, but they get confused due to same names of joysticks. Every time you start simulator, it loads same settings for both joysticks from one joystick. So every time need to assign axes and buttons again.

arthurlvjunk, 2021-07-17:
Hi,
It's actually fairly easy to do. You need to install Atmel Studio, open the solution file (MegaJoy.atsln). 
The product name is declared in the file usb_gamepad.c (look for #define STR_PRODUCT)

Note that you also need to change the product ID (a few lines below in usb_gamepad.c) so that Windows or your target software using those joysticks don't get confused. I recommend increasing the number to 0x82C2, 0x82C3, etc..
Make sure you are not using an already existing product ID (check devicehunt.com), otherwise Windows may try to install unwanted drivers.

Once you're done, clean and build the solution and it will generate a new hex file that you can flash your bootloader with.

One thing to consider as well is that Windows will buffer HID information and the updated name may not appear at first. You can unistall the driver in the device manager, but I managed to reset it by deleting the relevant entries in the windows registry below : Computer\HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaProperties\PrivateProperties\Joystick\OEM
Your experience may vary.

Hope this helps !
Arthur