AlanChatham/UnoJoy #50: Not a bug, I'm just missing something (probably obvious)

pr3sidentspence, 2022-10-30:
I can switch back and forth from arduino to joystick, no problem. And I can use the UnijoyArduinoSample and a Serial.println() to apply a function to remap an analog pin to what I need (from 226-32 to 0-255). I can move the axis and see that it's reading what I need through the serial monitor.

What I can't figure out is how to use that code as what's used to make the Uno a joystick. I tried replacing the UnoJoy\ATmega8u2Code\HexFiles\UniJoy.hex file with my own compiled from my edited sample, but I got an out-of-range error when running TurnIntoAJoystick.bat. 

I didn't expect it to work, but I also tried just TurnIntoAJoystick.bat after my edited sample on the board. My expectations were not unmet.

Any help would be appreciated.

manteka, 2023-09-11:
As with the Pro Micro, you can use Arduino Joystick library:
https://github.com/AlanChatham/UnoJoy/issues/56#issuecomment-1713638387

AlanChatham, 2023-12-21:
Hi! Sorry that I never got around to answering this last year, and I hope you resolved this problem, but in case it's something that you've still been working on, the way UnoJoy works is that you are writing code on the main Arduino chip (the ATmega328p) that's sending a specifically-formatted serial data stream to the USB communications chip on the Uno (the ATmega8u2). The normal ATmega8u2 firmware that comes with the Arduino shows up to your computer as a USB serial port converter, so the code in the ATmega8u2Code folder (UnoJoy.hex) is a firmware replacement that shows up to the computer as a USB video game controller instead of as a serial port, and turns the UnoJoy-formatted serial data from the main chip into controller button presses.

So, the workflow for UnoJoy is:
 - Code for your Arudino using the UnoJoy.h library
 - Upload that code to your Arduino working as a normal Arduino
 - Reset the Arduino into the communications chip bootloader mode by connecting some pins on the ISP header
 - Use the TurnIntoAJoystick.bat script to re-flash the communications chip
 - Unplug and plug back in, and it should show up as a joystick

To turn things back, it's:
 - Put the communications chip back in bootloader mode by connecting some pins on the ISP header
 - Run the TurnIntoAnArduino.bat script
 - Unplug and re-plug in!

closed 2023-12-21 by AlanChatham