AlanChatham/UnoJoy #58: ButtonBox request

Linrox, 2023-10-06:
Any chance of creating a 32 button box version for mega & uno

AlanChatham, 2023-12-14:
Sorry for the super late reply, but I'm not sure what you mean by a 32 button box? I'm ironically not super deep into custom controllers in the wild; is there a particular controller that you're thinking of? Currently, MegaJoy is an option for creating 32 button controllers, although it's also kind of goofy in that it creates two logical controllers as to have places for every one of the Mega's many, many pins.

Linrox, 2023-12-14:
Hey
I guess you have been busy, i didn't think i was going to get a reply. Thank you for replying.
Firstly i don't know if you can help. But thank you for listening.

MY GOAL
I am trying to create a button box for Elite Dangerous.
I don't think i will be adding a joystick, just buttons. 

So in using megajoy i don't know what code to use to send a signal so when a button is pressed(1 of 32) so the controller config software only detect 1 button pressed. I believe it is not print or write (i could be wrong)
And i want a little help in getting the code right. 
I have tried individual connected buttons and matrix style. both give the same result of either no key detected or multiple keys detected.

For example pushing a single buttons shows up as multiple buttons pressed. (leftDpad, LeftAnalog, UpAnalog, DownDpad, LTrigger,Akey,StartKey) all at the same time.

There are very few videos for linux button box's. So i am trying every option i can to learn

Thanks

AlanChatham, 2023-12-21:
Hmmm... I'm not super familiar with Linux, so I may be less helpful than I could be, but could you share the Arduino code you're currently trying to get working? I've also updated the MegaJoy firmware last week to fix a potential USB Product ID collision, so can you re-download the current MegaJoy/ATmega8u2Code folder contents and try that?