AlanChatham/UnoJoy #60: AtLibUsbDfu.dll not found.

clepee, 2023-12-16:
Hello, 

I have problems converting my arduino to Joystick, I use arduino uno R3 non generic board, so I use CH341SER driver. When I press the .bat the problem "AtLibUsbDfu.dll not found" appears.
[image]

AlanChatham, 2023-12-21:
Hi! My first guess says that this is a driver issue - have you run the UnoJoyDriverInstaller.bat file in the Drivers folder, and if so, did that give any errors?

echo1432, 2024-11-03:
I'll chime in here as I'm having the same problem.  
[image]

I'm not sure what I'm doing wrong, I've tried searching and am getting nowhere.  Any help is appreicated.

P8Enthusiast, 2025-09-06:
https://www.youtube.com/watch?v=KQ9BjKjGnIc&t=40s&ab_channel=MDGrein 
I know this post is old but this is a great video, you just need to install a driver