AlanChatham/UnoJoy #61: Hos to make a paddle controller for pong on pc

Marvan0305, 2023-12-29:
Hey can someone help me make a paddle controller with an arduino uno so i van play pong on my pc. 
Like this instructables page: https://www.instructables.com/USB-Paddle-Game-Controller/

Marvan0305, 2023-12-29:
I have 2 10kΩ potentiometer and 2 buttons.

AlanChatham, 2024-02-24:
Hi! Sorry this is as late as it is, but if you're still working on this, you could probably get away with just using the UnoJoyArduinoSample.ino code, connecting your potentiometer to 5v, GND, and the middle pole to A0 on the Arduino? Then the buttons connect between ground and whatever digital pins give you the buttons that you need.

I hope this helps!

Marvan0305, 2024-02-24:
Hi thanks for the anwser. I got the controllers working but the paddle is still moving when i stop rotating. I'm using a steam game called project:pong which is only has keyboard support. So i need to find another way to make paddle controllers for the game.

AlanChatham, 2024-02-24:
I wondered about that, if the controls were "paddle position is mapped to the joystick's position" like actual pong, vs. the apparent behavior of "moving the stick in one direction makes the paddle move that direction". 

You're probably going to be unable to make a potentiometer work in this case, since it's always going to be stuck in a physical range between it's minimum and maximum, and there won't be a sensible way to calibrate that against whatever position the paddle in the game starts at, and even then, it would be very difficult to exactly map the game paddle range with the physical position of the potentiometer, and you'd still get drift issues that would eventually make it so your on-screen paddle would stop before it hit the end of the screen, but you've moved the potentiometer to it's physical edge.

However, if you got a rotary encoder, you could easily make this controller, you'd just have to compare rotary encoder positions across each loop, and if the encoder position changed, you'd set the joystick to left or right depending on which direction the change was, and center the joystick position in the code if there was no change.

Hope it helps!