AlanChatham/UnoJoy #64: Can't upload any code to Adruino uno after turning back to Arduino using TurnIntoAnArduino.bat

Bluedriver01, 2024-03-09:
As title says after turning back to Arduino every attempt takes a few minutes and end with error "avrdude: stk500_recv(): programmer is not responding" does anyone know how to fix this. if you need any more info on issue feel free to ask.

AlanChatham, 2024-03-09:
I don't mean to be insulting, because I run into this too an embarrassing amount of times, but have you unplugged it and plugged it back in? It's also possible that it reconnected on a different serial port, so you should try those options in Arduino next. If those things don't work, is the system still seeing a joystick? Or is it somehow stuck in bootloader mode - check this by running TurnIntoAnArduno.bat and seeing if it shows any errors.

Let me know what you get!

Bluedriver01, 2024-03-10:
Firstable thank you for such fast responce i didn't expet that.

Bluedriver01, 2024-03-10:
Firstable thank you for such fast response i didn't  expect that

"have you unplugged it and plugged it back in"

Yes I did that, in addition to that I restarted my PC and also I plugged it into my laptop to check if the problem is with my PC or the problem is with Arduino and gives me same result on laptop so is the Arduino problem

"It's also possible that it reconnected on a different serial port"

well it was ealier conected to COM3 port and now is in COM4 but that also could cause problems on laptop? and if thats the case what should i do?

"is the system still seeing a joystick?"

No, it shows arduino UNO in device manager and in IDE also if computer would still see Joy I wouldn't use it in IDE because it doesn't show there (at least for me)

"Or is it somehow stuck in bootloader mode - check this by running TurnIntoAnArduno.bat and seeing if it shows any errors."

Ok this is copy most of command window contents

AT90USB82 - USB - USB/DFU

Device selection....................... PASS
Hardware selection..................... PASS
Opening port........................... AtLibUsbDfu: 3EB 2FF7 no device present.
FAIL    Could not open USB device.
ISP done.
Trying to re-flash for an Arduino Uno R3

ATMEGA16U2 - USB - USB/DFU

Device selection....................... PASS
Hardware selection..................... PASS
Opening port........................... PASS
Reading Bootloader version............. PASS    1.2.0
Erasing................................ PASS
Selecting FLASH........................ PASS
Blank checking......................... PASS    0x00000 0x02fff
Parsing HEX file....................... PASS    Arduino-usbserial-uno.hex
Programming memory..................... PASS    0x00000 0x00fd9
Verifying memory....................... PASS    0x00000 0x00fd9
Starting Application................... PASS    RESET   0

Summary:  Total 11   Passed 11   Failed 0

as you see there are no errors